bambulab/BambuStudio #107: Wall loops not changable

melfahn, 2022-08-09:
No matter what I set Wall loops to it stays at three when sliced. I want to increase it for strength, I usually go up to six for that, but nothing changes in the visual when sliced or when printed. 
[image]
[image]

YifanWuBambu, 2022-08-09:
@melfahn 
I tested the case but cannot reproduce the issue.
[image]

Please check whether it has object settings, which will override the global settings.
[image]

If it still does not work, please share the 3mf. Thanks!

melfahn, 2022-08-09:
thank you, that was the issue.  I'm going to share another my original project with you where I tried to  print part of the part with a 100% modifier.  in all of my tests trying to do this this leaves the outside layer a mess

closed 2022-08-09 by melfahn