bambulab/BambuStudio #109: Bambu Studio: Feature Request

kuljaden, 2022-08-09:
Hello Bambu Labs team,

Thank you for all that you've done so far! This is a feature request to add an option to export settings profiles. This should allow us to collaborate together to get perfect settings on our prints.

Thank you

randyre, 2022-08-09:
Agree!

YifanWuBambu, 2022-08-10:
Sure. We will add it to our plan list.

dereknheiley, 2022-08-12:
> Hello Bambu Labs team,
> 
> Thank you for all that you've done so far! This is a feature request to add an option to export settings profiles. This should allow us to collaborate together to get perfect settings on our prints.
> 
> Thank you

could you rename the issue title so that it's legible from the list of all issues on the previous page?

kuljaden, 2022-08-12:
> 

Hi Derek,

Renamed title to: Bambu Studio: Feature Request - Export Slicer profiles

Thank you

ChrisG-hub, 2022-08-12:
This is possible now, not elegant as a button 'save as', you just have to copy the *.json file of the profile you did tuned.
Just look into the C:\Users\cgarn\AppData\Roaming\BambuStudio\user\773006789\process folder, de numbered subfolder can have a different name. After that you can share the profile.

YifanWuBambu, 2022-09-13:
Please update to v1.2. You can export currently selected preset in File > Export > Export Presets

JamesD3d, 2023-04-04:
Please look into adding support for cancel object or exclude region