bambulab/BambuStudio #113: Feature Request: Axis labels and colors

sleeper-seth, 2022-08-09:
Other slicers helpfully color code the X | Y | Z labels to make it immediately obvious what coordinate is represented in the model view.

Current Bambu:
[image]

Cura Example:
[image]

Another small quality of life upgrade would be to label the X,Y,Z axis indicator when in the Move state (or maybe always?)

[image]

YifanWuBambu, 2022-08-10:
Thanks for the suggestion. We will optimize it in recent release.

closed 2022-08-10 by YifanWuBambu

YifanWuBambu, 2022-09-13:
It is in v1.2 version.

aa755, 2024-12-21:
the same could be done for the mirror tool:
[image]