bambulab/BambuStudio #115: Option to send to X1 or SD card in X1

VACIndustries, 2022-08-11:
While the printer is busy, or even if it is not; there should be an option to send the sliced file to the printer or SD card instead of only the two options to export or immediately print.

Allram, 2022-08-12:
That would be a great feature!

dereknheiley, 2022-08-12:
agreed, i use this all the time in klipper! to queue up the next couple of prints i want to slice after the current job is done

dereknheiley, 2022-08-12:
this seems related, or duplicates https://github.com/bambulab/BambuStudio/issues/61

YifanWuBambu, 2022-08-15:
It is a good idea. We have added it to your plan list.

Sir-Gavin, 2022-08-30:
this would be great

jl-p, 2022-10-22:
Agree. or just like my ultimaker queue it and let me press a button when i removed the current parts and start printing again.

And an option to print all buildplates in one project. with the same button to continue when the parts are removed.

YifanWuBambu, 2022-12-16:
This is included in v1.4.

closed 2022-12-16 by YifanWuBambu