bambulab/BambuStudio #117: Use STL or build plate + Project name in 3mf filename when exporting or printing

dereknheiley, 2022-08-12:
Currently if you have multiple build plates in a project, they all get sent to the printer as the same file name in the UI on the printer. Same problem if you "Export sliced file". You have to manually rename the file instead of having the default be STL or build plate + project name
[image]
[image]
.

YifanWuBambu, 2022-08-15:
We will optimize that in recent release.

dereknheiley, 2022-08-16:
Here are a few more.examples from slice files ok n the SD card, at least there is a preview picture 😄 
[image]
[image]

YifanWuBambu, 2022-08-17:
Thanks for the examples.

YifanWuBambu, 2022-09-13:
It is implemented in v1.2.

closed 2022-09-13 by YifanWuBambu