bambulab/BambuStudio #119: [Bug - v01.01.01.01] Print sequence by object with auto arrange could cause collisions

adnanklink, 2022-08-12:
I had a print fail from running into a printed object while printing in sequence. The printed objects were auto arranged on the build plate.

[gooose-collision.gcode.3mf.zip](https://github.com/bambulab/BambuStudio/files/9329581/gooose-collision.gcode.3mf.zip)

https://user-images.githubusercontent.com/32601617/184435232-ba2a868e-ed45-46e3-9aa5-adda633b2018.mp4

tangketan, 2022-08-15:
Please kindly provide the original 3mf. The file you attached is only gcode.

adnanklink, 2022-08-15:
[Sequence_collision_sample.zip](https://github.com/bambulab/BambuStudio/files/9334936/Sequence_collision_sample.zip)

Recreated it by adding the 5 items, changing printing sequence to by object, and hitting auto arrange.

tangketan, 2022-08-22:
Indeed it is a bug. We have solved it in Bambu Studio. Will sync the patch to github soon.

tangketan, 2022-09-01:
On the latest 1.1.1.4 this bug is solved.

closed 2022-09-01 by tangketan