bambulab/BambuStudio #120: Filament Flushing Volume menu button is not clear.

RobinWhitfield, 2022-08-12:
The icon to load the filament flushing volume menu is really unclear.
[image]

Firstly it's not clear that it's a button so many people miss that it even exists

Secondly it's unclear what the icon represents.

I'm not totally convinced that this is the right way to go design wise, but here's a free suggestion:
[image]

YifanWuBambu, 2022-08-15:
Thanks. We will optimize this in the next release.

RobinWhitfield, 2022-09-09:
They went super simple. I like it

[image]

closed 2022-09-09 by RobinWhitfield