bambulab/BambuStudio #121: [Feature Request] Clog detection or lidar layer check every N layers to detect clog

dereknheiley, 2022-08-12:
[image]
[image]

It would be awesome if the camera or lidar sensor could regularly check (maybe every 5 or 10 layers) check for a lack of Z height and warn / fail print because of clog.

sickerin, 2022-08-14:
+1. Would be really useful. This issue is more common than spaghetti for me.

tangketan, 2022-08-15:
This feature is under development. Will be released in a month or two.

tangketan, 2023-08-16:
After hundreds of hours of developing and massive testing of the "clog detection with lidar" function, I'm sorry that we decided not to publish the function in the recent period. There are mainly two problems:
1. A lot of false alarms. There are so many situations of printing objects and settings, and it is not easy to get a robust algorithm to handle all the situations. Our testing showed that it's very easy to trigger false alarms.
2. Lifetime of the laser can run out quicker than expected. The laser was designed for light usage, only a few minutes in each print. If we enable clog detection with Lidar, the laser will be used for a few hours per print, roughly 100 times more than designed.

We are still keeping this idea in mind, thinking of other ways to deliver this function. I'll keep this issue open until we final get a solution.

M1-Olsson, 2023-10-13:
@tangketan 

Maybe you can use the AMS to detect that the printer no longer use any filaments as a detector for clogs?