bambulab/BambuStudio #124: Only one wall on top surfaces setting is ignored

espressomaster, 2022-08-13:
Seems where walls are thin, only one perimeter is generated regardless of this setting:

[image]

whereas I would expect this top surface to essentially be only perimeter (3 wall loops selected), like the layer below:

[image]

SaltWei, 2022-08-15:
The topmost surface is always one wall as designed currently for better surface quality.
And it can't be influenced by the option "only_one_wall_top".

espressomaster, 2022-08-15:
> The topmost surface is always one wall as designed currently for better surface quality and no influence on strength.
> And it can't be controlled by the option "only_one_wall_top".

SaltWei, 2022-08-16:
Accept, will make the topmost surface also be controlled by the option "only_one_wall_top"

closed 2022-08-16 by SaltWei