bambulab/BambuStudio #125: Missing ability to see Printer Build-volume Boundaries

UncleJessy, 2022-08-13:
Most slicers will visually show you if an object is outside of the buildvolume of the printer. This is helpful for the user especially when working with multiple files on the buildplate and knowing which files are too tall to fit the print bed. 

Prusa Slicer [image]
Cura Slicer [image]
Simplify3D [image]
Bambu Studio [image]

YifanWuBambu, 2022-08-15:
We will optimize it in the next release.

lanewei120, 2022-09-09:
already added this logic in the newest released version v1.2.0

closed 2022-09-13 by YifanWuBambu