bambulab/BambuStudio #129: [Feature Request] Ability to send print files to the printer storage, so that they can be reprinted from the printer in the future.

SystemXploit, 2022-08-13:
I would love it if there were a way to manage the internal storage of the printer through the software/app. At minimum accessing the micro sd card if you have one plugged in. I have a lot of various prints that I need to reprint throughout the week. I currently use octoprint and when I need to print something I power on my printer remotely with a smart switch, use the app remotely and access the files that I have saved on the printer and reprint. I really expected this to have something similar.

I think it would be good if after the print when you get the popup asking if you want to reprint or tap success if it also gave you the option to save to local storage. I also think it would be nice if you had that option at the same time as you sent the print to the printer over the cloud, as well as had another tab in the slicer that let you organize your files properly.

UncleJessy, 2022-08-14:
1000% this! Even if it meant using an additional microSD card... but having a way to slice & wirelessly send the file to the printer. 

A great example of this working is how Raise3D Manages sliced files that are sent to the printer wirelessly

randyre, 2022-08-15:
I agree.

YifanWuBambu, 2022-08-15:
Yes. We will add this feature in recent release.

AbeFM, 2022-08-17:
Being able to reprint last file is nice, it would be better if that were ~10 files.
Related - be able to pick AMS slot to print from when reprinting.

YifanWuBambu, 2022-08-17:
> Being able to reprint last file is nice, it would be better if that were ~10 files. Related - be able to pick AMS slot to print from when reprinting.

Agree with you. They will be in our recent Studio & printer firmware release.

ACJUK, 2022-11-08:
+1 for this.  I just suggested in in the Facebook group and was told to come here to add it.  I too have come from Octoprint and a more Octoprint like experience for managing jobs would be really handy

AbeFM, 2022-11-08:
I can absolutely, positively live without a screen full of sample files. They are awesome for the first week and a half, but outside of the tool-handles, none of them are useful for long.

Perhaps you could "star"/favorite prints, and delete the built-in ones.

YifanWuBambu, 2022-11-09:
This feature will be included in the next release.

YifanWuBambu, 2022-11-09:
> I can absolutely, positively live without a screen full of sample files. They are awesome for the first week and a half, but outside of the tool-handles, none of them are useful for long.
> 
> Perhaps you could "star"/favorite prints, and delete the built-in ones.

The next firmware release will support to remove build-in samples.

closed 2023-02-22 by SaltWei

AbeFM, 2023-02-22:
Ideally, the last several prints - a screen full like the samples, but relevant, would be available. The ability to send to SD card is appreciate and makes a difference - but does not fully address what would be an awesome feature: Reprinting recent files from the printer without having to use the phone app.