bambulab/BambuStudio #130: [Feature Request] Multi-material painting using texture maps

jwalker55, 2022-08-13:
While the current painting tools are very good, I think it would open up more possibilities if you could control which AMS slot/tool is used based on a texture map. Obviously, the model you're slicing would have to be properly unwrapped. 

The way I would imagine this working is in the Bambu Studio settings, you could specify a texture map color for each AMS slot. Then, in the multi-material painting tools, you would have an option to use a texture where you can import the image and apply it to the model. During slicing, the slicer would determine which color to switch to based on the texture. When you paint your texture (Photoshop, Gimp, etc), you simply paint the parts that you want to use AMS slot 1, with whatever color you specified for the AMS slot 1 texture map color. You could use this to put logos and things on your model without having to actually make the logo part of the geometry. You could also make cool patterns or other things. Obviously, there would be some limitations around the texture, such as only being able to use solid colors, etc, so it works properly.

I imagine this might be a complicated feature to develop, but it's just an idea.

YifanWuBambu, 2022-08-15:
Thanks for the suggestion. It is very cool.
We will add it to our plan list, but may be not in recent release.

randyre, 2022-08-15:
Mosaic Canvas has this feature. It's a great idea. They call it stamping I believe.

jwalker55, 2022-08-15:
> Thanks for the suggestion. It is very cool. We will add it to our plan list, but may be not in recent release.

Awesome!

rkarlsba, 2023-01-08:
Seems IdeaMaker has something similar, but not for multicolour https://www.youtube.com/watch?v=ip21bJwRU3I

I'd really like this :)

nhnifong, 2024-02-22:
If I had a full color texture, say from a video game model, and I down sampled to a fixed 4-color palette, I could use dithering to create the illusion of more colors than I have available in my AMS, and it might even look good if your UV unwrap allocated similar resolution to all surfaces.