bambulab/BambuStudio #131: [Feature Request] Load Filament function (NON AMS Machines)

UncleJessy, 2022-08-14:
For machines that do not have AMS Attached, it would be nice if there was a LOAD filament option. 

Currently, you have to set the Extruder temp, load the filament, hit the extrude button multiple times

RobinWhitfield, 2022-08-26:
Potentially part of this request - presuming this happens, can the load feature have some way of overriding the temperature it does it at?

I don't want to load (or unload) POM at 250C...

YifanWuBambu, 2022-09-03:
@UncleJessy 
There is a problem that you need to push the filament into extruder AFTER temperature reaches the target value.

RobinWhitfield, 2022-09-03:
Can issue not be solved by user prompts?

Once a user has clicked the "load" button, it could move to the purge location, tell you "waiting for nozzle temperature", and then "please load filament" along with automatic extruder function once the sensor detects the insertion of filament. 
Then it can automatically purge a standard length, before wiping the nozzle.

YifanWuBambu, 2022-09-04:
@RobinWhitfield 
This may solve part of this issue. And the user still have to waiting for nozzle temperature.
Another problem is the purging length. Assume that both old and new filaments are PLA, but with different colors. The purging length from Black to White is about 3X of from White to Black.

RobinWhitfield, 2022-09-04:
Considering that before a print it purges a load anyway, even if the standard purge didn't clear it 100%, that likely wouldn't be a problem, and it's only doing it once per filament spool change so I wouldn't consider it a huge problem if it over-did the purge.

closed 2024-01-15 by SaltWei