bambulab/BambuStudio #132: Disable GapFill

PowerChaos, 2022-08-14:
hi,
i like to ask to enable a option to disable gapfill
in some cases it is usefull but in most cases it is not
it create a lot of retraction and small infills ( thats why it is gap infill ) on places it can just ignore
as you see the image , the gapfil alone takes longer then the innner walls , 

in prusaslicer it get disabled by puttin the speed on 0 , in bambuslicer it falls back to default speed

Thank you
Greetings from PowerChaos

[image]

SaltWei, 2022-08-15:
Could you show the printing case that gap infill has bed influence?

PowerChaos, 2022-08-15:
> But you opinion that gap infill causes more retraction and travel is right. For time saving, gap infill should be reduced, and currently we are evaluating “Arachne Engine” and variable line width.

based on my prusa and my 2 years playing around with the prusa slicer dit gap fill provide me a lot of trouble with certain prints ( default disabled in my prusa slicer ) 
i also got some prints where i need to enable it or the prints i made are useless ( no wall strength , mechanical prints ) 

i agree on technical stand point , if you print mechanical parts ( like holders ) then gap fill is needed to increase strength
if you print just funny objects ( a wobbly snake, a benchy ...  ) then gap fill is a waste of time as those do not even need any strength 
 
 same counts for the new "Arachne Engine"
 based on what i saw , it eliminate gap fill as it use fixed line instead, so it eliminate retraction/clogs and makes a variable line width, just like it can do for the Z height but then to eliminate gaps

 example of where it is needed
 
[image]

 example of wasted gapfill ( red is top solid infill  , purple is solid infill ) 
 
[image]
[image]
it apply gap fill to bridge the top layer ?
[image]

sidenote , just checked again with prusa slicer , they improved gap fill to, where bambu does gap infill prusa just put normal solid infill.
it is disabled by me as default.
[image]

SaltWei, 2022-08-15:
Could you share this model stl file so that we can check in detail? Thanks!

PowerChaos, 2022-08-15:
[laptop.zip](https://github.com/bambulab/BambuStudio/files/9336585/laptop.zip)
this zip contains the example files , the "neusje" is the one on top , the "laptop 0-3" is multi color and 'laptop single" is single color

SaltWei, 2022-08-15:
[image]
The left is prusa 2.5, default setting and disable the option of gap infill.

PowerChaos, 2022-08-15:
[AMC_Moon.zip](https://vpsdata.be/bambu/AMC_Moon.zip) is a other example of wasted gapfill ( 10% in that case ) with a lot of retractions
[image]

compared to what prusa does ( if gapfill  is enabled ) 
[image]

the problem are all those tiny dots that cause massive retraction and caused clogs for me in the past
so forced gap fill when gaps are XX size would be a nice solution , then it wont slow down or jump like crazy
also based on personal experience ( prusa ) it just does not fill up tiny dots anyway, it is to less material to fill up anything

sidenote, i also print at 0.16 layer height , first layer is 0.2 and then 0.16 for all other layers.

SaltWei, 2022-08-16:
[image]

The left is the latest released 01.01.01.01 BambuStudio. The right is latest internal BambuStudio which hasn't been released but will coming soon. The right uses arachne engine for internal narrow solid infill to replace ConcentricWithGapInfill pattern.

Gap percentage can be reduced, but solid infill increases.
No obvious time saving as long as the area should be fully filled to be solid.

But I still prefer the right one. Will coming soon.

PowerChaos, 2022-08-16:
To prevent confusion. This is all I can share as it is a prototype and not from me 
Pictures are taken when printed
When done on prusa they are smooth ( 100mm/s ) but that is with gap-fill disabled 
Maybe the new replacement will eliminate the need to disable gap-fill but as of today it still mess up some complicated prints 

Also as Sidenote. Gap-fill is taking 22% of the print and supports take 28%
[image]

[image]
[image]
[image]
[image]

SaltWei, 2022-08-17:
What's the model? Is this the model you shared in [AMC_Moon.zip](https://vpsdata.be/bambu/AMC_Moon.zip)?
I can print in Bambu Lab X1 Carbon to check whether to be such bad.

PowerChaos, 2022-08-17:
> What's the model? Is this the model you shared in [AMC_Moon.zip](https://vpsdata.be/bambu/AMC_Moon.zip)? I can print in Bambu Lab X1 Carbon to check whether to be such bad.

the files from the latest pictures are not shared on here , i can send it to you on discord ( or to @sirwill ) as i got permission to share it to you and him only for this purpose.

SaltWei, 2022-08-19:
> > What's the model? Is this the model you shared in [AMC_Moon.zip](https://vpsdata.be/bambu/AMC_Moon.zip)? I can print in Bambu Lab X1 Carbon to check whether to be such bad.
> 
> the files from the latest pictures are not shared on here , i can send it to you on discord ( or to SirWill on discordl ) as i got permission to share it to you and him only for this purpose.

Hi PowerChaos, SirWill shared the file to me yesterday. I sliced the model with 0.20mm Standard profile and Bambu PLA basic filament, and printed it on two Bambu Lab X1 Carbon machine. The quality is perfect for me. Pictures are shown below.

[image]

[image]

So I can't reproduce this issue. Is it possible to be caused by filament rather than slicing?

PowerChaos, 2022-08-19:
Thank you ,
They indeed looks nice.
i tried this with Esun ABS and with MakerFill ASA ( sliced on 0.16 ) 
both works great on the Prusa, 
but it is possible that the slicing is changed between different versions as when i put it in the slicer now i get again other value's then the screenshot i provided ( better value's ) 

i guess i going wait for the latest release and see how things go on from there
The main point of the gap fill remove was to reduce time as it fills in spaces it should not fill in
to prevent certain clogs ( the file that caused a lot of clogs was the laptop one, massive gap fill on the keyboards but seems fixed now ) or just because the gap fill takes as much time as the print itself ( 20%+ of time just to fill useless space ) 

in the end , if you can save 20% of print time or reduce it , then it is worth to disable it ( at least for me  ) 
Same as infill combination can be disabled or enabled to decrease print time 
even adaptive layer height can help in some cases to decrease print time

SaltWei, 2022-08-19:
Thank you.I will check again with ABS again. My Studio version is

hutje, 2022-09-03:
I do not see the need for this much gap fill and internal solid infill...
I have attached the 3mf file to check...
[BalloonDog.zip](https://github.com/bambulab/BambuStudio/files/9483230/BalloonDog.zip)

[image]
[image]
[image]
[image]

Bpositivedotdk, 2022-10-31:
I would also very much like to have the option to remove Gap infill. In many cases it adds a lot of print time....
On this lamp, its over 5h of 23h total. 

[image]

CNCXPROUK, 2022-11-02:
It may also be a good idea to have the option to disable it for the first couple of layers.  I’ve had issues where on small prints the speed of the printer doing a gap fill drags the part off the plate because the surface area of the part isn’t enough to keep it bonded.

ab3dgeek, 2023-03-07:
I would also love to have the option to turn off gap infill pls!
I print tiny objects sometimes and it's totally not needed, it is a lot of extra time, a ton of extra retraction, creates stringing and all the extra movement knocks delicate parts over causing hours of printing to fail.

hutje, 2023-03-07:
> I would also love to have the option to turn off gap infill pls! I print tiny objects sometimes and it's totally not needed, it is a lot of extra time, a ton of extra retraction, creates stringing and all the extra movement knocks delicate parts over causing hours of printing to fail.

The fork of SoftFever does allow to disable the gap infill. And you can specify the threshold of the length to ignore smaller gaps.

ab3dgeek, 2023-03-07:
> > I would also love to have the option to turn off gap infill pls! I print tiny objects sometimes and it's totally not needed, it is a lot of extra time, a ton of extra retraction, creates stringing and all the extra movement knocks delicate parts over causing hours of printing to fail.
> 
> The fork of SoftFever does allow to disable the gap infill. And you can specify the threshold of the length to ignore smaller gaps but keep the necessary gap fills

thanks for the tip, I know and use it ;)
but for example last weekend it just stopped working on my m1 macbook as it is not signed and this time malware protection just straight killed it, so as much as I like softfever and I do, I can't rely on it (this is my business as well as my hobby).

hutje, 2023-03-07:
Apple make life unnecessarily difficult for developers. For end-users it is a blessing, for the developers a curse...

LidorBaum, 2023-04-29:
Is it planned to be added to the slicer? I think it's important for some situations.

QingZhangBambu, 2023-08-01:
In classic mode, you can filter the small gaps by using Filter out tiny gaps. To use that parameter, you need to open develop mode. Hope that will be helpful.
[image]

[image]

[image]

QingZhangBambu, 2023-08-01:
This function is added on version 1.7, you can update your software to get that.

closed 2023-08-01 by QingZhangBambu