bambulab/BambuStudio #133: [Feature Request] Filter out speeds that are not used for better color contrast

dereknheiley, 2022-08-14:
Currently, if a filament sets a lower Max Volumetric Speed, the scale and color gradient for the max speed is still based on the values in the process speed. Even though you can clearly see most of the speed colors are never used during the print.

The speed visualization below would be much more useful if the highest value was the max actual speed used in the sliced output. Then create the color gradient based on that range.
[image]

YifanWuBambu, 2022-08-15:
Hi @dereknheiley, could you share me your .3mf file?

dereknheiley, 2022-08-15:
> Hi @dereknheiley, could you share me your .3mf file?

sure, here you g0
[compactyl-v6-alpha-bottom-plate-left.gcode.3mf.zip](https://github.com/bambulab/BambuStudio/files/9334964/compactyl-v6-alpha-bottom-plate-left.gcode.3mf.zip)

YifanWuBambu, 2022-08-17:
@dereknheiley 
I found the gap fill speed is about 300 mm/s. It caused this issue.
[image]

dereknheiley, 2022-08-18:
Ah, my mistake! I didn't zoom in that far. 

Still, if there's nothing between 100 and 300, it seems like this would be an area for improvement. It looks like the supports aren't actually running at 150mm/s because of the max volumetric speed. So the actual used speeds are 1~100, and 300?

Even after implementing arachne with variable line width. It seems like there will still be some points in the model that are filled with rapid tiny extrusions like this.

RobinWhitfield, 2022-08-26:
The speed displayed should be the speed the printer is going to print at, taking into account the max volumetric speed.

Changing the view to "flow" will show you how fast the supports and infill are printing in mm^3/2

One thing I have noticed is that as the calibration is always long enough to hit the maximum volumetric speed if you have small models (and when you turn the calibration off it's *after* slicing so it's always there), there can be quite a small variation in colours on the models themselves. It might be nice to be able to (like the tick boxes for different line types), adjust a slider on the flow and speed settings in order to only show parts of the model within that flow/speed range.

YifanWuBambu, 2022-08-31:
@dereknheiley @RobinWhitfield 
It is actually an area to be improved. We have added this to the plan list.

YifanWuBambu, 2022-10-12:
Hi @dereknheiley ,

We will display the speed value of each path in the preview.
[image]

YifanWuBambu, 2022-10-21:
Added in v1.3.0.12.

closed 2022-10-21 by YifanWuBambu