bambulab/BambuStudio #136: [REQUEST] automatically move calibration pattern to an empty part of the build plate

tsdexter, 2022-08-15:
Can a feature be added to automatically move the calibration pattern to any available spot that is large enough on the build plate to enable prints with large perimeters but lots of empty space elsewhere. For example a ring with 246mm inner diameter and 256mm outer diameter. 

https://github.com/bambulab/BambuStudio/issues/32 mentions a pause feature to remove the calibration pattern before carrying on with the print but it would also be ideal to not have to pause and remove it if you have the space on the build plate elsewhere.

tangketan, 2022-11-09:
It's a bit difficult to add this feature. As you can see from the Studio, the calibration pattern is defined in start gcode, it's impossible to change its position or orientation for now.

tsdexter, 2022-11-09:
ok, thanks. Would still be nice to have someday, but I'm sure the 'pause after calibration' to remove the lines before printing will suffice for now

SaltWei, 2023-02-22:
"pause after calibration" has been supported.

closed 2023-02-22 by SaltWei