bambulab/BambuStudio #138: mac Bambu Studio v01.01.01.01 crashes when WAN network changes to offline

dereknheiley, 2022-08-16:
how to reproduce on mac OSX Monterey 12.3

1. start mac Bambu Studio v01.01.01.01
2. click Device tab at top
3. click play in camera pane
4. observe video displays
5. turn off WAN cable internet modem, keep LAN wifi router online
6. observe can still communicate with LAN devices like klipper UI & klipper camera in fluidd/mainsail
7. observe bambu studio crashes with OSX dialog showing stack trace & repoen application
Expected: bambu studio should communicate with camera over LAN, or turn off camera feed without crashing

StoneLiBambulab, 2022-09-01:
We tried to reproduce it and it didn't happen, please help  upload the crash report when it appears again.

SaltWei, 2023-03-02:
Please let us known if still exist on v1.5.0 public beta.

closed 2023-03-02 by SaltWei