bambulab/BambuStudio #141: Ability to center an object

randyre, 2022-08-18:
Right click context menu item to center an object. Needed especially when adding multi part objects, the ability to center the object to get them to line up.

YifanWuBambu, 2022-08-19:
@randyre 
When loading multi-parts object, please drag them into Bambu Studio together then it will popup a dialog for importing them as a whole object.
[image]

If it still solve your problem, please help share your model to us. Thanks!

randyre, 2022-08-20:
If I drag and drop and it aligns the parts then later I move a part, how do I get them to line back up again without either centering tools or merge tools or something like that.?

klawrow, 2022-08-21:
@randyre have you tried using the "Arrange all objects" button?

[image]

UncleJessy, 2022-09-03:
> @randyre have you tried using the "Arrange all objects" button?
> 
> [image]

That does not reliably work
I have a large object that fits within the buildvolume of the printer
[image]
When I use the Arrange function it moves it off the buildplate and I have to manually move it to the center
[image]

tangketan, 2022-09-03:
Try set "brim type" to "no-brim" and see if the object is arranged correctly.

klawrow, 2022-09-03:
Hmm interesting. I'll have to try a bigger model.

UncleJessy, 2022-09-03:
> Try set "brim type" to "no-brim" and see if the object is arranged correctly.

Yeah that actually worked. I'd still like the option to center my print regardless of the object size in relation to the buildplate like in other slicers

YifanWuBambu, 2022-09-03:
@randyre @UncleJessy 
Yes. We may add that feature.

YifanWuBambu, 2022-10-21:
Added in v1.3.0.12.

closed 2022-10-21 by YifanWuBambu