bambulab/BambuStudio #144: [Feature Request]

PowerChaos, 2022-08-19:
Dear Developers
this is just a small suggestion
as most printers that use ABL or that kind system got a locked in place system for the bed is a visualizer not needed ( as you can not adjust it anyway )
but the Bambu has a bed with screws, so it can manual be adjusted and that can mess up the bed leveling
this also means that it can mess with the bed once the springs start to loose there tension/quality
I am sure that this has a lot of advantages over a fixed bed 
but it would be nice to have a certain way to see how level the bed is and that adjustments can be made

there are 2 ways this can be done
a easy way is that the system just checks the data and the skew it has ( 0.2 on left and 1.8 on right -> that's not correct ) and then suggest how much degree you need to turn the knobs ( 360 degree to left on right knob -> recheck -> suggest or continue ) , also known as Semi Level Assist

Second way is to provide a visual system in the slicer , as i am sure it got already data send to it from the AMS so it would not be to hard to get the mesh send to
then you can manual adjust or relevel or do whatever you like to do as that is the case with octoprint and all other printers that use a level system ( hooked to octoprint ) , also known as Manual Level Assist

it is just a suggestion to remove the need to manual level or to have a way to at least check if the bed is still level and that it does not need to jump a lot
how flatter the surface level , how better and cleaner your prints are , how less it need to correct with Z offset on each mesh point
that is also the reason why you need to manual level your bed first before you do a auto level ( on most printers ) or why it is fixed to the frame and can not be adjusted

YifanWuBambu, 2022-08-23:
@PowerChaos That's a great idea. We have added this to our plan list.

ChrisG-hub, 2022-08-23:
Cool.

SaltWei, 2023-08-16:
This need work from firmware side. @sodachen22 Please evaluate