bambulab/BambuStudio #146: [Feature Request] Have button to display filament vs. build plate guide in slicer

dereknheiley, 2022-08-19:
I know this seems trivial, but for anyone who's just getting into printing, or using a different filament for the first time, having to go and find the chart of what build plate will work better/worse for your filament would really go a long way.

[image]

YifanWuBambu, 2022-08-23:
@dereknheiley That's a good idea. We will add a link to at the "Bed type" text. Clicking it will jump to the bed type intro web.

ptyork, 2022-08-26:
I'll just add to this a suggestion that having a checklist (ensure proper build plate, open glass top, open door, ensure glue stick, etc.) could be added to the final print dialog.

YifanWuBambu, 2022-09-14:
Added in v1.2.
[image]
[image]

closed 2022-09-14 by YifanWuBambu