bambulab/BambuStudio #147: PA-CF Filament Profile is missing as a default system present when adding .6 nozzle via setup tool

accetturo, 2022-08-19:
The PA-CF Filament Profile is missing as well as many others, but Bambu labs should cover this as it's a manufactured filament and you recommend a .6 nozzle on the sales page. Can anyone share it and can we get it added to an update?

The reason that Bambu X1C is such an amazing printer is it just works when using all Bambu parts and filaments, so having the right profiles for the various nozzle sizes, especially if it's your recommendation, is critical to user success.

YifanWuBambu, 2022-08-23:
The PA-CF filament preset for 0.6mm nozzle will be in the next release.

accetturo, 2022-08-23:
Thank you, Can we get the rest of the profiles included as well, especially the supports?

YifanWuBambu, 2022-08-23:
Absolutely!

SaltWei, 2022-09-19:
Has been included in latest version 1.2.0.9

closed 2022-09-19 by SaltWei