bambulab/BambuStudio #148: "Random" seam position missing

tokamac, 2022-08-20:
In PrusaSlicer, there are 4 settings for the seam position:
- **Random**
- **Nearest**
- **Align**
- **Rear**

But in BambuStudio, there are only 3 settings:
- **Nearest**
- **Aligned**
- **Back**

So "Random" has disappeared.

I do not consider this a "feature" but a bug. Random seam position can led to a fuzzy surface in some occasions, but for specific models that have no back vs front face (an object with rotational symmetry for example) that can be manipulated and looked from any angle, random dots are sometimes preferable over an unsightly very visible line.

Please consider bringing this feature back into BambuStudio.

YifanWuBambu, 2022-08-23:
Okay, we will add it in the next release.

SaltWei, 2022-09-07:
Done.

closed 2022-09-07 by SaltWei