bambulab/BambuStudio #149: Antivirus software reports it's a trojan virus

stofancy, 2022-08-20:
I'm using [火绒](https://www.huorong.cn/) in my laptop, it's reporting that the `BambuSource.dll` is `Trojan/Emotet.mv`, during BambuStudio first run:

[image]
病毒名称：Trojan/Emotet.mv
病毒ID：577E70823C93A22A
病毒路径：C:\Users\ztmds\AppData\Roaming\BambuStudio\plugins\BambuSource.dll
操作类型：修改 
操作结果：已处理

进程ID：50868
操作进程：D:\Software\Bambu Studio\bambu-studio.exe
操作进程命令行："D:\Software\Bambu Studio\bambu-studio.exe" 
父进程：C:\Windows\explorer.exe

病毒名称：Trojan/Emotet.mv
病毒ID：577E70823C93A22A
病毒路径：C:\Users\ztmds\AppData\Roaming\BambuStudio\plugins\backup\BambuSource.dll
操作类型：修改 
操作结果：已处理

进程ID：50868
操作进程：D:\Software\Bambu Studio\bambu-studio.exe
操作进程命令行："D:\Software\Bambu Studio\bambu-studio.exe" 
父进程：C:\Windows\explorer.exe

## System Info
设备名称	ANKAIHANG-Y9000X
处理器	Intel(R) Core(TM) i7-10875H CPU @ 2.30GHz   2.30 GHz
机带 RAM	32.0 GB (31.8 GB 可用)
系统类型	64 位操作系统, 基于 x64 的处理器
BambuStudio Version: 01.01.01.01
Antivirus Version: 火绒 5.0.69.5

Allram, 2022-08-20:
I get no vrius warnings on bambu-studio.exe from release 01.01.01.01.
Might be a false positive on your software?

https://www.virustotal.com/gui/file/78334c3146e309b04de18b5a7e4c13f9012e6a93ff9c5d2647d0c7a7580f24d6

stofancy, 2022-08-21:
Thanks @Allram , anyway, I've reported false warning issue in the antivirus software community: https://bbs.huorong.cn/thread-109606-1-1.html

lanewei120, 2022-08-23:
> I get no vrius warnings on bambu-studio.exe from release 01.01.01.01. Might be a false positive on your software?
> 
> https://www.virustotal.com/gui/file/78334c3146e309b04de18b5a7e4c13f9012e6a93ff9c5d2647d0c7a7580f24d6

Thanks @Allram for providing this

the plugin files seems also ok:
https://www.virustotal.com/gui/file/423626d7af74d546589687a3942479393cf23ee5ecbac78449aba3c8ca926602
https://www.virustotal.com/gui/file/25972390daf36858d25db0322e3be62889d5f46399c31afe91160a64adf7c9ec

stofancy, 2022-08-28:
@Allram @lanewei120 , got confirmation with Huorong, this is a false warning, they will fix it by upgrade on their side. Issue is closed.

closed 2022-08-28 by stofancy