bambulab/BambuStudio #150: Feature: Non-planar slicing / printing

psiberfunk, 2022-08-22:
Since the dimensions of the print head are precisely known for the X1 series, it should be possible , within the limits of not being able to tilt the print head, to implement non-planar slicing, which, in particular, can significantly improve surface finish on surfaces that might otherwise show a "topo" map effect. 

This is an ambitious feature, something that no other mainstream slicer has implemented meaningfully, but there are modern implementations out there: https://xyzdims.com/2022/03/26/3d-printing-slicing-with-non-planar-geometries/ .

Could be a neat unique-to-bambu feature to advertise.

psiberfunk, 2022-08-29:
I agree , this would certainly be more useful on a printer where the 3 point bed tilting was available, but I believe that in many prints the top surface finish has areas like you describe that are actually pretty important to have smooth.

gabebear, 2023-03-26:
Non-planar printing also has potential for cutting down on the number of material swaps needed. 

Any non-planar slicer is going to need to take into account the nozzle running into existing  structures above what is currently being printed. This same logic could potentially cut down on the number of filament changes needed.