bambulab/BambuStudio #151: [Feature request]: Velocity Painting / Tattooing

psiberfunk, 2022-08-22:
It would be super cool, and unique, if BambuSlicer were to bring the concept of velocity printing into a more mainstream audience that was accessible via an intuitive GUI and compatible with modern slicers.  Velocity painting is basically utilizing the different visual appearance of plastic printed at different speeds to "watermark" an image onto a surface.  Here's some examples of what it can look like (swiped from someone else.. the Evee isn't really a good example, but others show it well):
[image]

An implementation of the velocity slicing technique shown here https://www.velocitypainting.xyz/blog/ and implemented in perl here:  https://github.com/MarkWheadon/velocity-painting .  There's a basic app to show how it might be visualized here: https://www.velocitypainting.xyz/gui/index.html 
 
**How it would work**
Ideally you'd load in a grayscale PNG to map onto the surface, as with the GUI, and be able to manipulate the mapping in an intuitive way by "dragging" the image around in the 3D slicer view.  Similarly, you'd be able to scale in X/Y the image and tweak how the overlay object works.   This would in theory work like Microsoft's 3D Builder app when "embossing" a stencil or image into an STL mesh, from a UX perspective.

YifanWuBambu, 2022-08-23:
Yes. This feature is in our plan list.

psiberfunk, 2022-09-11:
Note @YifanWuBambu  that this feature is different than just embossing text on an object.. it specifically is using the technique of altering velocity.  I mention this because it looks like there's an upcoming text embossing feature which is a little different than this.

YifanWuBambu, 2022-09-11:
@psiberfunk 
Yes. This feature has not been implemented yet.