bambulab/BambuStudio #154: Feature Request: Current Layer

YyAoMmIi, 2022-08-22:
Both the Printer and Slicer list printing progress as % and as time remaining.
Request would be to add layer count. Say there 200 layers. So it would it it as layer 115/200

YifanWuBambu, 2022-08-23:
Hi @YyAoMmIi , in what senario do you need the layer progress?

YyAoMmIi, 2022-08-23:
This is more a fun feature. Say one knows to expect when slot 3 color start to be used at layer 34, so they start to watch print more  when print is currently on layer 30.

YyAoMmIi, 2022-08-23:
In Addition: It would help if someone wanted to make changes at specific layers. 
1) say making a temp tower that more for BL Carbon
See convo by Ciprian in the discord server
https://discord.com/channels/970163353542606888/970163355379695681/1011513029294305281
2) If someone wanted to do a multi material print that uses more than 4 material but have only 1 ams. knowing specific layer helps them know when to swap out the material.

ChrisG-hub, 2022-08-23:
Been asked in https://github.com/bambulab/BambuStudio/issues/41
They have added an option in a previous version but it disappeared again in the latest version.

dereknheiley, 2022-08-26:
> Hi @YyAoMmIi , in what senario do you need the layer progress?

The slicer doesn't show percentages when you look at the preview and drag the slider through layers. And the printer doesn't show layers as it prints, only percentages. Adding current layer to the printer progress bar seems like the behaviour everyone is expecting to see based on every other 3d printer in the world.

YifanWuBambu, 2022-08-31:
@YyAoMmIi We @ChrisG-hub  @dereknheiley 
Thanks. We have added this feature to the plan list.

dereknheiley, 2022-09-09:
> Thanks. We have added this feature to the plan list.
> Currently, you may insert a pause at a specific layer for filament change.

If M117 support was added in #225 then you could display the current layer without any other UI work

**Layer change G-code**
`M117 layer:{layer_num+1} Z:[layer_z] ;output height for nerds `

brunic, 2022-10-11:
add one more scenario: insert something when printing specific layer. Like I printed something including cylinder magnets which should be wrapped in the printing part. I have to stand and count which layer is printing and pause printing when I need to add the magnets if without this feature.

candre23, 2022-10-12:
This is actually a useful feature that I've been missing.  Maybe it's just because I'm used to it from every other printer I've owned, but I do make note of what layer potentially-problematic features start at, so I know when to keep an eye out for problems.  It's not something you need to know all the time, but there are certainly times when it's useful.

timothyzalusky, 2023-02-07:
[https://wiki.bambulab.com/en/x1/manual/X1-X1C-AMS-firmware-release-history Added support for displaying the current layer number/total number of layers during printing](https://wiki.bambulab.com/en/x1/manual/X1-X1C-AMS-firmware-release-history)

Issue can be closed

closed 2023-03-02 by SaltWei