bambulab/BambuStudio #156: AMS Filament Type after Filament Runout Sensor

DrakeLewis, 2022-08-23:
This is a visual bug, but after the filament runout sensor is tripped and new filament is loaded the AMS display contains a "?" as if it is unsure what filament is loaded. It clearly does not seem to have any effect on the actual print though.
[image]
.

YifanWuBambu, 2022-08-23:
Hi @DrakeLewis ,
To make it more clear, please help confirm which one is the visual bug.
#1. AMS display contains a "?" after old filament runout and new filament is loaded 
#2. During actual print, the behavior int #1 does not happen

DrakeLewis, 2022-08-23:
#1

YifanWuBambu, 2022-08-31:
@DrakeLewis 
If the newly loaded filament is not from Bambu Lab (without RFID), actually the print does not know its material type and color.

MarcoJPereira, 2022-09-02:
In addition,  If you pause the print when the printer ask you to remove the filament etc... when the new filament will be loaded, the printer will go where it stopped and purge a blob right on top of the prints before going back to the "purge station" and returning to the print. If you pause/resume two times, you will have two blobs.

By the way, as the length of the filament between the sensor and the nozzle is known, why not purging the old filament when the AMS is running out of filament instead of asking the user to pull it?