bambulab/BambuStudio #160: [Feature request] Raft layers

reify3d, 2022-08-24:
Please consider to implement a raft layer feature in BambuStudio. This increases the probability of success for prints that have multiple small footprints all over the bed. A brim would not work because:

1. The total surface area covered by the brim is much less compared to a raft
2. The post-processing needed to clean up each individual brim is much higher than just peeling the print off the raft layer

YifanWuBambu, 2022-08-26:
@reify3d 
Could you share me your model?

reify3d, 2022-08-26:
The KS backer who experienced this issue is on the Discord, and has been notified.

wuyifanfancy, 2022-08-28:
Thanks for the information. If possible, please share us the STL or 3MF.
We will add raft feature in the plan list.

RobinWhitfield, 2022-09-08:
I would also be interested in a raft feature, for very warpy filaments that don't have good bed surface adhesion (specifically POM). 

Combined with the ability to pause and load new filament it would permit say a PLA raft to be used for POM to be printed on top of.

finsdotsurf, 2022-09-18:
With a raft option I can fit my large fin prints next to each other. Raising the middle object in this screenshot by about 15 layers is all I need. For now, gonna see if I can export a rafted version of it from PrusaSlicer.
[image]

BATEMAN87, 2022-09-25:
I would much like the raft option as it will be best suited for engineering parts. With a raft, elephants foot would be completely eliminated and surface finish would be ideal as opposed to textured build plates

YifanWuBambu, 2022-11-09:
Raft will be included in the next release.

YifanWuBambu, 2022-12-16:
This is included in v1.4.

closed 2022-12-16 by YifanWuBambu