bambulab/BambuStudio #161: [Feature request] Manual Layer Height Adjustment

nhguy2257, 2022-08-24:
While Adapted Layer Height it great and does the calculations fairly well. I would also enjoy having the option to adjust manually. So the whole print is .2 but layer after a certain height would be .08 for better finish.

accetturo, 2022-08-24:
Simply3d has an option like this I believe, would be useful for sure

nhguy2257, 2022-08-25:
> Simply3d has an option like this I believe, would be useful for sure

You can also do it in PrusaSlicer which Bambu is based off off... So hopefully an easy integration.

cdsmakestuff, 2022-09-12:
I just added another "feature request" for the same idea but a little more descriptive and how it is implemented in PS and SS...
https://github.com/bambulab/BambuStudio/issues/258

SaltWei, 2023-08-16:
Already support height range. Thanks for suggestions

closed 2023-08-16 by SaltWei