bambulab/BambuStudio #162: Adaptive Layer Height doesn't scale Tree support options

nhguy2257, 2022-08-24:
While having Adaptive Layer Height enabled, Tree supports will not scale so they fuse with the part and is impossible to remove.

tangketan, 2022-08-26:
Would you please share the 3mf?

nhguy2257, 2022-08-26:
[Deathstroke.txt](https://github.com/bambulab/BambuStudio/files/9433080/Deathstroke.txt)

Just rename .3mf

tangketan, 2022-09-01:
It's a bug in old version. Try the latest version 1.1.1.4, it's fine on my side.

closed 2022-09-01 by tangketan

nhguy2257, 2022-09-01:
> It's a bug in old version. Try the latest version 1.1.1.4, it's fine on my side.

Ok, great... Where is the latest 1.1.1.4? I can not find it anywhere.

tangketan, 2022-12-05:
@nhguy2257 Sorry for the late reply... I meant the bug was fixed 1.1.1.4, so any version greater than that should be fine.  Try the current latest release （1.3.1.1）.