bambulab/BambuStudio #163: Tree Auto Supports unusual behavior (printing in mid air or with limited support

accetturo, 2022-08-24:
Hello, I've ran into an issue that wasted an entire roll of Support W with PLA Basic. Basically, the Tree supports started building in thin area or with nearly no underlying material (building on the very edge of a wall with maybe a mm coverage)

I don't have the files and I can't replicate it, but there is an error with the logic on some geometry. I'd recommend inspecting the tree supports and using Tree Support with Infill option until it's resolved.

tangketan, 2022-09-01:
Indeed single wall hollow structure in tree support isn't strong enough for certain cases. Known cases include using Silk PLA on tall objects. We haven't tested using Support W on support base. Usually it's used on support interface. Seems  Support W is similar to silk PLA from your case.
Yes, it's a good suggestion to enable Tree Support with Infill option to keep it strong.  But we are concerned it may be too difficult to remove. We are still working on it, trying to find a good infill pattern balancing strength and removal ability.

tangketan, 2022-09-03:
By the way, "printing in mid air" as a bug in previious versions, and should not be seen in the current latest.

YifanWuBambu, 2022-10-21:
Fixed in v.1.3.0.12.

closed 2022-10-21 by YifanWuBambu