bambulab/BambuStudio #166: Fahrenheit options in the Printer and Slicer

accetturo, 2022-08-24:
It would be nice to have the ability to change to see fahrenheit on the printer for things like temp controls, chamber temp, bed, etc. I haven't checked the slicer, but it would be nice to have it in slicer as well

dereknheiley, 2022-08-26:
> It would be nice to have the ability to change to see fahrenheit on the printer for things like temp controls, chamber temp, bed, etc. I haven't checked the slicer, but it would be nice to have it in slicer as well

Honestly curious: does any filament manufacturer provide specs only in fahrenheit?

psiberfunk, 2022-08-29:
I’m in the USA , and I have never ever thought about printing in Fahrenheit .  I’m not sure what the value of this is ? 3D printing temperatures are always in degrees Celsius

SaltWei, 2023-03-02:
Evaluated to be won't doing. Please let us known if still necessary.

closed 2023-03-02 by SaltWei