bambulab/BambuStudio #167: Camera Failed To Load (0)

accetturo, 2022-08-24:
Getting errors "Camera Failed To Load (0)" after running long prints for PA-CF. Chamber isn't too hot, 40C, so I don't think it's overheating. The only way to fix it is to restart the printer.

lanewei120, 2022-08-29:
it seems not be a studio issue, but some issue on the printer

Is there any information shown on the printer's screen?
and also could the printer respond to the touch screen's event at that time

if it is reproducible，please help to take a picture of the printer's screen when it happens next time

> Getting errors "Camera Failed To Load (0)" after running long prints for PA-CF. Chamber isn't too hot, 40C, so I don't think it's overheating. The only way to fix it is to restart the printer. Fails to load in Slicer, Mobile App and on Printer.

brunic, 2022-10-13:
I think you're right. It's the printer issue because both the slicer and iOS app cannot connect the printer camera after a long printing(not sure how long but this issue happens everyday). No matter what filaments I use. The printer screen won't show anything wrong. Restarting the printer will fix this issue.

> it seems not be a studio issue, but some issue on the printer
>
> Is there any information shown on the printer's screen? and also could the printer respond to the touch screen's event at that time
>
> if [...]

sodachen22, 2022-10-29:
We reproduced the issue. It is a software issue related to camera service. It will be fixed in next firmware release

SaltWei, 2024-01-19:
Does this issue still exist？