bambulab/BambuStudio #168: Reprint issues picking filament

randyre, 2022-08-25:
When picking an object from the printer to print that has been previously printed we get the option to change AMS slots. This works as designed.

The problem comes that after changing AMS slots and print completes, and a reprint is picked it goes back to the default slot that's in the GCode and doesn't allow a change at that point. 

Correct behavior should be to default to the last used slot with an option to change it regardless if it is a print from a file (saved to the printer or SD Card) or a reprint after a print completes.

YifanWuBambu, 2022-08-26:
The behavior will be optimized. The device will let user to select AMS slots again when reprinting.

YifanWuBambu, 2022-10-14:
Fixed on firmware v1.1.0.0

closed 2022-10-14 by YifanWuBambu

randyre, 2022-12-04:
The problem is back. Reprint does not prompt for the AMS slot. Having to re-upload just to change colors.

YifanWuBambu, 2022-12-05:
@randyre 
What's your firmware version?
We may notify the firmware guys to check it.