bambulab/BambuStudio #169: Cannot open saved project 3MF file in PrusaSlicer

jstaylor11, 2022-08-25:
Cannot open a saved project 3MF file in PrusaSlicer.  I would think these would be compatible and conform to the standard 3MF format.  The exact error I get is "Error (Invalid 3MF format) while parsing '3D/3dmodel.model' at line 17".  Now that error is from PrusaSlicer, but it is because of the saved file from Bambu Studio in an invalid 3MF format.

Thanks for all the hard work you are putting into this!!!

deftdawg, 2022-08-25:
Did you try with SuperSlicer also?  If not, give it a try and see if it works.

lanewei120, 2022-08-26:
we had supported prusa 3mf in v1.1.1.1, the geometries can be loaded in Bambu Studio.
could you upload your 3mf for us for a try?

> "Error (Invalid 3MF format) while parsing '3D/3dmodel.model'

lanewei120, 2022-08-26:
Just to clear why we add this logic:
we found when saving a large model, the save time sometimes is slow, and it is caused by the big 3dmodel.model file;
then we split the 3dmodel.model into serveral model files,
each one refers to a single model object in the project.
[image]
we use the tbb to do the writing in parallel which can saves lots of writing time.

enricoturri1966, 2022-08-26:
> Currently Prusa doesn't support this

Actually, PrusaSlicer is not the only application unable to load the 3mf files generated by BambuStudio, as you can see from the attached images:
[image]
[image]
[image]
[image]
[image]

Same applies for Blender

lanewei120, 2022-08-26:
the 3mf saved by BambuStudio uses '3MF Core Specification' and '3MF Production Extension';
We will add an entry to export the 3mf only using '3MF Core Specification',  then other slicers can use it directly

dereknheiley, 2022-08-26:
Add idea maker to the list of slicers that can't open the 3mf files from bambu

closed 2023-03-02 by SaltWei

garnos, 2023-08-29:
I don't think this was resolved. Please let me know if that's not the case. I commented over here:

Rusty105, 2023-10-17:
> I don't think this was resolved. Please let me know if that's not the case. I commented over here: [prusa3d/PrusaSlicer#10718](https://github.com/prusa3d/PrusaSlicer/issues/10718)

Nope, as of October17, 2023 Bambu .3mf files do  not open in other slicers.

What is the point of a 'standard' if no one is following?

Rusty

bambu123, 2023-10-17:
> Nope, as of October17, 2023 Bambu .3mf files do not open in other slicers.
>
> What is the point of a 'standard' if no one is following?
>
> Rusty

There are several advantages to using the 3mf Production extension to manage slice projects:
1. When saving files, various sub-files can be generated by multiple threads to improve efficiency; while traditional solutions may save large-scale projects for more than one minute
2. Support backing up projects by part, with one sub-file per part; backed up sub-files can further improve project preservation efficiency
3. Support model data encryption

lanewei120, 2023-10-17:
Hi @Rusty105 

[image]
you can use this function to export a bambu 3mf which can be opened by other slicers

Let us explain more why the 3mf saved in the normal way by Bambu Studio can not be opened by other slicers:
1.  The 3mf from Bambu Studio follows two 3MF specs: [3D Manufacturing Format](https://github.com/3MFConsortium/spec_core) and [3MF Production Extension](https://github.com/3MFConsortium/spec_production/blob/master/3MF%20Production%20Extension.md)

as @bambu123 said above, we use [3MF Production Extension](https://github.com/3MFConsortium/spec_production/blob/master/3MF%20Production%20Extension.md) for our optimization

2. currently other slicer can not load our 3mf are just because they don't support the [3MF Production Extension](https://github.com/3MFConsortium/spec_production/blob/master/3MF%20Production%20Extension.md)

we have also submitted patches to these two slicers(cura https://github.com/Ultimaker/Cura/pull/15761, https://github.com/Ultimaker/libSavitar/pull/48; prusa https://github.com/prusa3d/PrusaSlicer/pull/10808) 
if they can accept these patches, our 3mf can also be opened by them;

3.  furthermore, the 3mf spec doesn't define the 3d print parameters' format
even the 3mf can be opened by other slicer, only the geometry data and color data can be loaded