bambulab/BambuStudio #170: Loading .3mf files locks filament colors

randyre, 2022-08-25:
Load a previously saved project and it won't let me change the filament slots in the AMS to what's currently in the device, just what was loaded when the project was saved.

The expected behavior would be to allow modifications, or even better to ask when loading the project if you want to load the original AMS filament slots (then you could go into the AMS and insert the proper colors) or to ask if you want to use the currently loaded slots.

YifanWuBambu, 2022-08-28:
@randyre 
Do you mean in the device page or "Send print" dialog? I tried both but did not reproduce this issue.
[image]
[image]

Could you share me your 3mf and give a screenshot where the problem happened? Thanks!

randyre, 2022-09-11:
The problem appears to be from loading a .3mf file that was saved in another slicer, and it only sees it as Gcode, not as a project file, there is no way to go back to the prepare tab to change things. I must have thought it was a bambu file.

closed 2022-09-11 by randyre