bambulab/BambuStudio #172: Ironing Bug

moebis, 2022-08-26:
Ironing simply doesn't work. Either top most or all top surfaces is sporadic. It does some sections and not others. Even tried copying the PS settings at 15% flow rate and 0.1 spacing. See example here: [https://imgur.com/7Mk5Pbx](https://imgur.com/7Mk5Pbx)

Cosmo991119, 2022-08-29:
could you provide your model?

moebis, 2022-08-29:
> could you provide your model?

https://www.printables.com/model/51074-fast-operation-vise

QingZhangBambu, 2022-09-02:
> > could you provide your model?
> 
> https://www.printables.com/model/51074-fast-operation-vise
> 
> also this bug was found on the Mac version.

the problem is solved and would be updated in the next release

SaltWei, 2022-09-07:
Just bug of gcode viewer. Fixed

closed 2022-09-07 by SaltWei