bambulab/BambuStudio #173: Video from printer camera playing in Bambu Studio needs optimization

bluecobras09, 2022-08-27:
During prints, I have Bambu Studio open and watching via the built in camera.  Sometimes as much as the whole bottom half of the screen will pixelate.  In discord I was told others have the issue.  If they watch the video via Bambu Handy, there are none of these artifacts so prevalent in Bambu Studio.  I see the same.  In Bambu Handy, video is perfect.  Bambu Studio. as much as half the time of the print, the bottom of the video is pixelated to some degree.

bambu123, 2022-08-31:
Does it appear when the picture is very complicated? I think the per-picture buffer is not big enough for the H264 IDR frame.

bluecobras09, 2022-08-31:
I really do not know how to determine if the picture is "very complicated".  There could be nothing even on the build plate and it will happen.  The print head could be stationary.  Others appear to not have the issue.  So, not really sure.

bambu123, 2022-09-01:
Please help to do the following to fetch a video dump for us:
1、Download GraphStudioNext from https://www.videohelp.com/download/graphstudionext64-0.7.3.2.exe
2、Open BambuStudio & play
3、When playing, right-click the white area under the video, and the video URL will be copied to clipboard
4、Paste the URL somewhere and append "&dump=1.h264" to it
5、Open GraphStudioNext, from menu File > Render URL, play the above-modified URL
6、Then click the play button & let it play for a while until you see the pixelated bottom
7、Send us the 1.h264 & 1.info files that would be dumped in the same folder where you open GraphStudioNext

bluecobras09, 2022-09-02:
Ok, in the process of trying to get a dump of these, I have run into a very strange problem.  I completed the 2.5hr print with only one glitch that happened so fast I didn't have the dump happening.  The print finished with no issues.  So I started a second print.  This is were things went sideways.  The bed did not heat and the flow test which is normally at the front of the bed was in the middle.  So I power cycled the unit.  Now it is having the pixelation issue.  Even when the unit is doing nothing.  I did get a dump and will send it.

bluecobras09, 2022-09-02:
Ok, I attempted to open a ticket in the support system for this.  Even though it appeared as though I was opening a new ticket, it just attached the information to the ticket I opened earlier as I am now having an issue where the bead will not heat.  I uploaded the dump to that ticket.  I just attached it to Ticket No: G_220902400000.

bambu123, 2022-09-02:
From the dump, I found the problem is that the h264 frame is too big.  We'll update BambuSource.dll in the next release. Thank you!

bluecobras09, 2022-09-02:
That is good!  At least it turned out to be something easily identified.

YifanWuBambu, 2022-09-13:
@bluecobras09 
In v1.2, if you start the camera feed, it will auto-resume after swithing back to the device page.