bambulab/BambuStudio #174: [Feature request] Power Down After Print

MJWMJW2, 2022-08-27:
I notice that the printer consumes around 24-25 watts idling after print.  Could there be an option to soft power down or be able to add some kind of standby after x minutes in printer firmware?

RobinWhitfield, 2022-08-28:
This has been discussed a few times in the Discord.

The issue as I see it is how the printer knows where it is once power is resumed.

An auto-home could only be done in the X and Y directions, as the build plate cannot be assumed to be empty.

The only solution I can really come up with is that on resume the printer could ask you to confirm (either on the touch screen or via the app) that the print bed is installed and empty, and then it can do its homing routine.

RobinWhitfield, 2022-08-28:
On a related note though, would it be possible to have a location to log issues and requests for the printer firmware? Even though the firmware is closed source, perhaps a repository with just an issues log for it could work?

This issues page is really only for the slicer.

YifanWuBambu, 2022-08-28:
> I notice that the printer consumes around 24-25 watts idling after print. Could there be an option to soft power down or be able to add some kind of standby after x minutes in printer firmware?

The low power mode is in already in our schedule. Thanks for the suggestion.

isaaxa, 2022-08-30:
> > I notice that the printer consumes around 24-25 watts idling after print. Could there be an option to soft power down or be able to add some kind of standby after x minutes in printer firmware?
> 
> The low power mode is in already in our schedule. Thanks for the suggestion.

The option to set idle times to power down the motors would be wonderful. Or at the very least allow the ability to use M18/M84, in the End Gcode to turn motors off after a print. I've tried and it doesn't seem to work.

timothyzalusky, 2023-01-13:
I think this ticket should be closed, power draw now down to 6w after print timeout.  Refer to [changelog](https://wiki.bambulab.com/en/x1/manual/X1-X1C-AMS-firmware-release-history)

"12. [System] Supported turning off the motor when printing is completed or the printer is idle for over 20 minutes. The idle consumption is reduced from 25w to 6w."

RobinWhitfield, 2023-01-14:
While I agree that dropping the power consumption to 6W is an improvement, in modern standby terms this is pretty insane.

Fundamentally I don't know how low things can go with the printer design as it is, but I'd like to think that this can be improved further.

The PSU used is rated at around 90% efficiency, and <0.3W no load, so there's quite a decent amount of power being lost somewhere else.

The distributed nature of the control boards may limit what can be done, but only the AP board needs to remain somewhat awake in this sleep state with limited communcation to the outside world.

MJWMJW2, 2023-01-14:
Agreed with @RobinWhitfield, 
The 6watt is likely keeping the wifi stuff alive.  Ideally give two configurable power save states
After x minutes - low power mode (aka 6 watts)
After y minutes - go to standby < 0.5 watts

Times should be configurable within certain parameters, 20 minutes is far too long in my opinion - should be able to be immediate after cooling has finished.

RobinWhitfield, 2023-01-14:
Even keeping WiFi alive should be possible with a lot less than 6W. I think it's keeping *everything* alive, just now switching off the motors entirely.
A couple of slicer versions ago they already made the change to reduce the motor current after a print which if I remember correctly took it down to around 18W on idle.

I can see some people wanting to keep the WiFi active at all times to enable remote wake, so I'd be happy seeing standby at <1W as an option, with a clear note that it will disable WiFi after this time and will require the machine to be poked to wake it back up.

RobinWhitfield, 2023-01-14:
Can anyone confirm whether the rear fan remains on in this low power mode? That's primarily blowing near the heatsink for the motor driving circuitry, so if it's still on it should be possible to turn it off.

MJWMJW2, 2023-01-14:
Not sure which fan was on, but certainly making a noise. 

Unit was sat at 8.5 watts average (230v). Left for a couple of hours to measure.

RobinWhitfield, 2023-01-14:
Quite a substantial "idle" power consumption then, and quite a way off 6W.

The MC board fan is the one that's always on, but it's controllable so there's no technical reason why they can't disable it on sleep - it certainly shouldn't be needed to cool anything down.

Does anything else obviously non-essential remain on? Toolhead light? Chamber light?

Merlwynd, 2023-01-14:
I guess I don’t understand the reasoning for Bambu to spend any more time on this (at least for now) vs the many other requests. If one is worried about low 6w consumption for long idle times or possible component wear, why not just turn the printer off? I have a smart plug and can do this remotely from anywhere. Printer boot time on power up is fast and not an issue….

MJWMJW2, 2023-01-14:
> Quite a substantial "idle" power consumption then, and quite a way off 6W.
>
> The MC board fan is the one that's always on, but it's controllable so there's no technical reason why they can't disable [...]

Chamber light was off.  I don't remember if toolhead light was on.

Just considering some of my AV gear, this has similar two step power save, and the ability to over ride with allowing network control, which uses a couple of watts extra.  Still they are using much less than than the printer.  Even so I already automated this to cut the power once the main fan stops after printing (around 27 watts), still it would be nice to be controllable, even just to be able to send an MQTT message, then I can automate from there rather than monitoring the power usage.

[image]

RobinWhitfield, 2023-01-14:
> I guess I don’t understand the reasoning for Bambu to spend any more time on this (at least for now) vs the many other requests. If one is worried about low 6w consumption for long idle times or [...]

I'm not saying Bambu should be spending lots of time on this right now compared to other things (like fixing the motion planning bugs), but I also don't think this Github request should be closed as complete.

Merlwynd, 2023-01-14:
ok...but just wondering the reasoning vs just shutting printer off? I mean if even WiFi shuts off to keep wattage at absolute minimum, I don't understand why one wouldn't just shut off the printer?

RobinWhitfield, 2023-01-14:
You could say that about almost any device.

When there are thousands of them that people will leave on (after prints, and probably just generally more often now that the standby power is less) there's no reason for the power consumption to be as low as possible.

Many people don't turn off their paper printers - they leave them on 24/7.

Merlwynd, 2023-01-14:
True enough...I guess from an environmental footprint it makes sense as lots of folks would do as you suggest. I tend to simply turn all my printers off when not in use...then again, I am a hobbyist and don't run a print farm for business (although in that case the printers aren't likely idle for long anyway)....

MJWMJW2, 2023-01-15:
> 

Firstly, at least many of us in EU have become used to certain devices that can only use 0.5 watts in standby*, so in general it is not needed to turn things off, maybe you can say we are lazy, or maybe we just assume devices don't zombie power when not being used anymore.

You mention paper printer - just checked my inkjet, which is on wifi, it is using <1 watt in this state.  You also make the point about "thousands of people will leave on" > this is exactly the point here; so 8 watts or whatever is no biggie for me, but when you multiply by thousands of these devices, then every other device that does this, it is significant amounts of power that could be better used, or indeed not produced in the first place.  Add on that caps in power supplies don't last forever, so you are saving planet in many ways, and avoiding an earlier device failure.

Secondly, if I am doing a long 3d print maybe it finishes when I am not around, or I am asleep, I'd rather save the few watts until I can physically turn the power off**

*Note this doesn't apply to some devices like TVs or amplifiers if you change settings to keep them always on a network etc, but even then they are typically 1-2 watts.

**but for me, I don't worry about this, as most devices like this are connected to power monitoring sockets, which will kill the power once they drop to a certain wattage - in the case of bambu, below 27 watts for >10 minutes

Even with all EU stuff making sure to reduce power consumption, my house is taking 300 watts average always.  The only critical things on are fridge, freezer, networking and NAS (around 200 watts average).. Somewhere is >100 watts being wasted for all those power adapters and power supplies doing nothing more than slightly heating the house 99.9% of the time.