bambulab/BambuStudio #175: [Feature Request] Implement interface shells option

GuyH77, 2022-08-28:
Please implement the interface shells option as found in Prusaslicer. This option ensures a smooth top interface layer, which improves print quality and also strength. This setting is used when printing multi-colour. 

[image]

In this example. A door sign. The top layer would benefit massively from a single monotonic layer and not lots of regions.
[image]
next layer
[image]

GuyH77, 2022-09-15:
This really needs adding, this is a simple QR sign and the base should just be an easy print. It doesn't need this interface layer.

[image]

GuyH77, 2023-08-02:
@YifanWuBambu any chance of this been added? Prusaslicer has this function and it makes such a difference to surface quality. Another example. This top layer doesn't need the gaps, it degrades the surface finish and I'm sure is slower. Any filament saved it negligible. I just need the option to ensure the best quality possible. 

[image]
[image]

SaltWei, 2023-09-04:
@BambooBugFixer Please add back this option in 1.8 version

GuyH77, 2023-11-10:
@SaltWei I don't see this in the new 1.8 beta? Please, this function will make my signs look so much better.

SaltWei, 2023-11-10:
@QingZhangBambu Please keep following this ticket

GuyH77, 2023-12-11:
Will this make it in to 1.8? It's not in Beta 3. I really need this functionality. Also, now you have the makerworld signs addon it will be better with this function. Please let me know if I can help test. Thanks.

GuyH77, 2024-03-25:
Still not fixed in 1.9 Beta.

SaltWei, 2024-03-26:
@QingZhangBambu @XunZhangBambu Please don't forget to add this in next beta

XunZhangBambu, 2024-06-26:
this option has been implemented , discussion under #2906

closed 2024-06-26 by XunZhangBambu