bambulab/BambuStudio #177: [feature request] View and interact with the LCD in Bambu Studio

josephpauljohnston, 2022-08-28:
If the software allows remote operation, let's go ahead and go the whole way. We should be able to view what's being displayed on the LCD at any time, anywhere. There are codes, interactions, and messages that pop up on the LCD that are not reflected in Studio, and no way to work with those messages. If we can operate SOME of the printer remotely, we should be able to operate ALL of the printer remotely.

RobinWhitfield, 2022-08-28:
I think being able to view exactly the same interface remotely would be totally unnecessary, but there are clearly messages and notifications that should be provided more readily on the mobile app and Bambu Studio.

I seem to remember quite the discussion about this and I thought there was an issue logged already, but I can't now find it.

There isn't for example any need to try and replicate the model picker page on a mobile app or Bambu Studio though.

YifanWuBambu, 2022-09-03:
@josephpauljohnston  @RobinWhitfield 
Notification and Error messages will be added in the next release.