bambulab/BambuStudio #179: [bug] Bambu Lab Studio 1.1.1.1 crashes while being launched on a secondary display

danielclasen, 2022-08-28:
**Description**
There is currently a bug preventing Bambu Lab Studio to start on a secondary Display. It happens when you move the Bambu Lab Studio Window to a secondary Display, close it while its still on your non-primary Monitor and then try to re-open it. It will just freeze until it gets the typical white-ish overlay for a non responding App and Windows reports that the App has crashed. Rendering the User to be unable to use the Slicer anymore.

**Steps to reproduce:**

1. Open Bambu Lab Studio
2. Move the Window to another Display
3. Close Bambu Lab Studio
4. Try to open Bambu Lab Studio once again
5. Feel the pain

**Environment:**
OS: Windows 11 Version	10.0.22000 Build 22000
Bambu Lab Studio: 1.1.1.1

**Temporary Workaround**

1. Disable all other Displays in Windows Display Configuration (or power them off/ unplug them)
2. Start the Slicer 
3. Re-enable/start/plug all your Displays again
-> It will automatically launch on your primary Display working as intended (as all other Displays are disabled/powered off/unplugged while you start the Slicer)

Mikerium, 2022-08-31:
+1 am also having this issue.

ChrisG-hub, 2022-08-31:
Can you both give me you Windows version (build number), and used video card brand?
The more technical info the better.

Do you have the latest bios, hardware drivers, ect...?

Mikerium, 2022-08-31:
Sure thing, 
Win 11 Home, version: 10.0.22000 Build 22000
Processor: 12th Gen Intel(R) Core(TM) i9-12900K
Mobo: ROG STRIX Z690-A with Bios: American Megatrends Inc. 0807, 12/3/2021
GPU: EVGA Kingpin 3090ti with driver version: 516.94

I'm able to get the app to work again if I uninstall and remove user settings and then reinstall. But just like the OP if I were to move the app to a different screen and then try to open it afterwards it'll hang here: 
[image]
and won't ever load, giving me a spinning cursor if hovered.

lanewei120, 2022-09-09:
already fixed in the newest released version v1.2.0

closed 2022-09-13 by YifanWuBambu

danielclasen, 2022-09-24:
This does still happen in the current v1.2.0.9 of Bambu Studio :/ 

Running:
* Windows 11 Pro (22H2) 22621.521 
* with Windows Feature Experience Pack 1000.22634.1000.0
* AMD Adrenaline 22.9.1 (AMD Radeon RX 6900XT)

and 3 Displays:
* Main: 1x 34" 3440x1440 
* Top: 1x 29" 2560x1080 (landscape mode)
* Left: 1x 29" 1080x2560 (portrait mode)

Whenever I close Bambu Studio on the Top or Left Display I can't open it up again. It will freeze while launching as described in the posts above.

richardhartless-wk, 2023-10-09:
+1 this also happens on Mac's running
running macOS 14 (but this was also happening in macOS 13)
Bambustudio 1.7.6.92
all monitors run through hub and are managed via Displaylink

walterwongbbl, 2023-10-10:
> +1 this also happens on Mac's running running macOS 14 (but this was also happening in macOS 13) Bambustudio 1.7.6.92 all monitors run through hub and are managed via Displaylink

Did all of this happen when Studio was moved from one screen to another?

walterwongbbl, 2023-10-10:
Hi. @richardhartless-wk 
If you only have one screen, is everything normal?

vad3rr, 2023-10-11:
if you're having this problemo, and its stuck on a side monitor, just set that monitor to your primary monitor, move it to your main, set that back to primary and enjoy

arnaudlvq, 2024-08-24:
Still happening on Windows, 1.9.3.50 :(

ajquick, 2024-11-28:
Still happening.  When having two monitors attached and you resume from sleep, Bambu Studio is just a white screen. I have to kill the task and hope that whatever I had that I was working on is still there.

JohnSmithCYM, 2024-11-29:
@ajquick Can you offer more information about your screen, I have tried about 10 times with my laptop(2K, 150% Ratio) and sub screen(2K, 100%) and it didn't crash or be blank.My OS is Win11 and I tried unplug subscreen or make laptop into sleep mode.

ajquick, 2024-11-29:
Yes. Windows 11, two monitors one ACER one LG. 

What typically happens is when my monitor goes to sleep it completely disconnects. Then upon awaking from sleep or the monitor awakes from sleep it reconnects as if I just plugged my monitor in. Under either scenario the Bambu Studio can be a white screen.

I'd suspect that this is some kind of a hardware initialization thing. Perhaps if you have it on your secondary monitor and that monitor goes to sleep (disconnects) Bambu Studio is probably still trying to draw the program on that now non-existent monitor. It should perhaps detect if the monitor still exists and move to a monitor that does currently exist.

Galliyummy, 2025-01-22:
somebody should reopen this issue, i have a broken laptop screen and i am literally unable to use the slicer due to this issue.

Calisto88, 2025-02-13:
> somebody should reopen this issue, i have a broken laptop screen and i am literally unable to use the slicer due to this issue.

Probably you fixed this already 

But for some folks that still find this post a workaround is set first your second monitor as the main screen and then you can run it.