bambulab/BambuStudio #180: resume print after spaghetti detection resets chosen filament

MNT22946, 2022-08-29:
For a print job spaghetti detection is turned on. On spaghetti occurrence the print job should pause.
Starting a print with a given filament slot - let's say 4 - will start printing.
When the spaghetti detection is triggered (this happend without any real failure during print whats how ever?!) the print job stopped correctly.
By resuming the print job the filament of slot 1 will be taken to print further!

HobinBambu, 2022-09-03:
@MNT22946 when spaghetti detection stop the print job, did you press "resume" to continue or stop and restart the print job?could you please upload log refer to this wiki https://wiki.bambulab.com/en/x1/troubleshooting/upload-log
thanks.

MNT22946, 2022-09-03:
I pressed resume when I realized that the printer is using the wrong
fillament I stopped the print job.
I can't upload the proper log files. According to the wiki log files should
be generated right after the problem  occurs and I have had many prints
after that.

HobinBambu, 2022-09-03:
@MNT22946  so the wrong filament (slot1)were used from the beginning of this printing?If so, this is a AMS mapping issue, which will be fixed in a next firmware.