bambulab/BambuStudio #181: Request - Timelapse "with" toolhead option.

RobinWhitfield, 2022-08-29:
It's super simple to add the code for this into the layer change g-code so that it just takes a photo and doesn't move the printhead, but it would be great to have an option in the slicer to make it easy for users to enable/disable it.

A formalised version of the else code, with a selection option:
`{if timelapse_no_toolhead}
G92 E0
G1 E-[retraction_length] F1800
G1 X65 Y245 F12000 ; move to safe pos
G1 Y265 F3000
M400 P100
M971 S11 C10 O0
G92 E0
G1 E[retraction_length] F300
G1 X100 F5000
G1 Y255
{else}
M971 S11 C10 O0
{endif}`

Sirwill1968, 2022-08-29:
I agree, timelapse with toolhead by not adding additional time, or wipes or timelapse tower would be great.  Zero impact to print quality, and no increased time.  Or bangs to clean the nozzle.

YifanWuBambu, 2022-09-01:
We will schedule this in recent release.

RobinWhitfield, 2022-09-09:
This is now present as "traditional" timelapse mode in the latest 01.02.00.04

closed 2022-09-09 by RobinWhitfield