bambulab/BambuStudio #182: [bug/feature request] slicing error: hinge gap is fused/closed by slicer (due to g_config_slice_closing_radius in libslic3r)

danielclasen, 2022-08-29:
**Description**
Currently the Bambu Lab Slicer 1.1.1.1 is too aggressive while closing x/y gaps during slicing resulting in delicate CAM/print-in-place mechanics to fail (hence the models are actually fused during printing). Take this example of a hinge where the actual .stl has gaps all around the hinge mechanic. The upper side of the hinge is being sliced/fused together with the rest of the model while the lower part is keeping their gap as intended (red circles in screenshot):

[image]

Or the same model, just a few layers higher. The upper part of the hinge is still being sliced together with the base of the model but also the left side is now also being merged in. The lower side of the hinge is still keeping its gap as designed in the model:

[image]

This is the .stl for reference, it was created by [Vit Budina over at printables.com ](https://www.printables.com/model/137770-print-in-place-dump-bucket-module/files)
[Dump_basket_MODULE.zip](https://github.com/bambulab/BambuStudio/files/9447550/Dump_basket_MODULE.zip)

The original setting in Slic3r controlling this slicing behaviour is called `Slice gap closing radius`. However this is not actually configurable due to:
https://github.com/bambulab/BambuStudio/blob/ba83769030c3fe98c0b564106967dec4fbc32efb/src/libslic3r/libslic3r.h#L90

and thus it is then used in here:
https://github.com/bambulab/BambuStudio/blob/cddce117c94d3fabb3f89b4412d30f287b55c463/src/libslic3r/PrintObjectSlice.cpp#L139

resulting the print to actually fail in the end. Well, at least hinging or other CAM mechanics fail due to that slicing error.

**Proposal/Feature Request**

I do see the point in making Bambu Slicer easy and straight forward to use shipping with fine tuned profiles resulting in awesome prints (and oh boy, did you deliver on that! I really mean this :heart:). However as there is currently no way for users to change a constant like this `g_config_slice_closing_radius` you might want to consider at least allowing those detailed libslic3r settings to be adjusted in the json profiles directly. As a last resort so to say, while you can elaborate internally on even more refined profiles to ship the slicer with.
And yes, I know that the gap and tolerances in this model are quite tight, but lets face it. You build a truly remarkable machine that is indeed capable of handling tolerances on this scale. If it just boils down to a slicing issue it would be a shame to leave that performance on the table 😁

WeiTaoSJTU, 2022-09-07:
accept and will add this setting back

SaltWei, 2022-09-07:
Accept. Will add it back on UI

closed 2022-09-07 by SaltWei