bambulab/BambuStudio #183: Fix model missing from Mac version

chendo, 2022-08-30:
The "Fix Model" option in the context menu of a model does not exist on Mac, but does on the Windows version.

GuzziRaz, 2023-09-10:
I can't picture what non-portable Windows code that would be? Surely Windows doesn't have any native support for STL whatsoever?

I found out how to fix files in Fusion 360 though, not too difficult but a bit tedious.

lanewei120, 2023-09-11:
> I can't picture what non-portable Windows code that would be? Surely Windows doesn't have any native support for STL whatsoever?
> 
> I found out how to fix files in Fusion 360 though, not too difficult but a bit tedious.

it is from windows sdk
https://learn.microsoft.com/en-us/uwp/api/windows.graphics.printing3d?view=winrt-22621

GuzziRaz, 2023-09-12:
I hear PrusaSlicer can fix model under macOS now, could you not just merge that code?

lanewei120, 2023-09-13:
hello @GuzziRaz 

we have tried it before, the fix model under macOS is not a really fix, just some simple process logic to stl
if you find it, could you give a picture of the fix model entry on prusa under macos?
we will do more investigation then

thanks

AtomicRPM, 2023-11-02:
Any update on this?