bambulab/BambuStudio #188: Memory Leak in 01.01.01.01 Windows Version

LegacyOfBob, 2022-09-01:
I've had Bambu Studio open for 10 days or so doing various prints on my X1C. The memory usage has been steadily increasing and is now close to 26GB.
[image]

LegacyOfBob, 2022-09-01:
I should add that I was doing lots of multicolor prints via the AMS over the past week or so.

aquatacizem, 2022-09-01:
I can concur.
I have also had issues trying to slice complicated multi-colour prints, where slicing never progresses past 5% (slicing mesh) even after more than 12 hours. Bambu Studio just keeps chewing up memory, seemingly working but with no progress.

LegacyOfBob, 2022-09-02:
Hello,

I am printing these models: https://www.thingiverse.com/thing:2525047

Each one of the landscape tiles is it's own print and takes 8-12 hours and a lot of color changes. I have left Bambu Studio open and after each set is complete I save the .3mf of the project, then I open a new project and "transfer" the settings over. I change colors or whatever else is necessary and print the next file. I left Bambu Studio open in this way for 10 days or so and probably printed ~140 multicolor models for around 120 hours of printing.

lanewei120, 2022-09-09:
already fixed a memory leak in the newest released version v1.2.0

please have a try again in your cases

closed 2022-10-14 by YifanWuBambu