bambulab/BambuStudio #189: Request - Bambu Studio reload capability

Vmtr, 2022-09-01:
In Bambu Studio, add either an ability to detect change in an STL file and prompt to reload, or to right-click a model in a project and select a "reload model" option. This would be helpful when re-printing a model as its design is evolving.

YifanWuBambu, 2022-09-03:
@Vmtr 
This is in our plan. We will add "reload" feature in recent release.

closed 2022-10-21 by YifanWuBambu

DzzD, 2023-01-21:
Hi, this issue should not have been closed, I think OP (as I would like to) was thinking more of an automatic detection, not a manual reload.

When a source file stl is modified a popup should ask to reload, reload should apply all transformation made in Bambu studio (obviously when applicable, not painting or other unrecomputable features)

samy, 2024-02-21:
I agree with @DzzD : this feature exists in Cura and other software and is very useful. You export your STL file on the same file and the slicer should offer auto detection of changes