bambulab/BambuStudio #191: Tool head hits objects on the plate if time lapse and sequential printing are enabled

CptHungarian, 2022-09-01:
There are actually two separate bugs in this case:
1. If time lapse and sequential printing are both enabled, the tool head hits the previously printed objects on its way to the excess chute (to take the photo for the time lapse video).
2. The slicer suggests to add time lapse wipe tower to the plate, but in sequential print order the tower is printed out separately.
[image]

YifanWuBambu, 2022-09-03:
@CptHungarian 
This bug has been fixed.
In sequential printing, prime tower will be disabled.

RobinWhitfield, 2022-09-04:
@YifanWuBambu, does that mean that timelapse without toolhead has also been disabled for object-by-object printing, or can it now calculate a path to avoid previously printed objects/print them in an order front-back?

YifanWuBambu, 2022-09-05:
> @YifanWuBambu, does that mean that timelapse without toolhead has also been disabled for object-by-object printing, or can it now calculate a path to avoid previously printed objects/print them in an order front-back?

1. Does that mean that timelapse without toolhead has also been disabled for object-by-object printing?
Yes. And we will add another timelapse type (snape a picture after a fixed interval), which can be used in the by-object printing.

2. Can it now calculate a path to avoid previously printed objects/print them in an order front-back?
In by-object print, if we still use "timelapse without toolhea", each object needs an individual prime tower.
That will take a lot of area of the build plate, and the video effect may not be very good.

YifanWuBambu, 2022-09-13:
Fixed in v1.2.

closed 2022-09-13 by YifanWuBambu