bambulab/BambuStudio #192: BambuStudio quit unexpectedly v.01.01.01.01 - MacOS 12.5.1

mapaolini, 2022-09-02:
BambuStudio quit unexpectedly v.01.01.01.01 - MacOS 12.5.1

Was rotating in Prepare.

System Problem report:
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               BambuStudio [78734]
Path:                  /Applications/BambuStudio.app/Contents/MacOS/BambuStudio
Identifier:            
Version:               01.01.01.01 ()
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               504

Date/Time:             2022-09-01 23:25:40.1601 -0500
OS Version:            macOS 12.5.1 (21G83)
Report Version:        12
Anonymous UUID:        2D0920B0-FCDF-D3C0-2755-3E00ACA845D7

Sleep/Wake UUID:       EA43417A-A195-4532-B716-28224C532BDC

Time Awake Since Boot: 340000 seconds
Time Since Wake:       24 seconds

System Integrity Protection: enabled

Crashed Thread:        0  bambustudio_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000030
Exception Codes:       0x0000000000000001, 0x0000000000000030
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [78734]

VM Region Info: 0x30 is not in any region.  Bytes before following region: 140737486426064
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      VM_ALLOCATE              7fffffe29000-7fffffe2a000 [    4K] r-x/r-x SM=ALI  

Kernel Triage:
VM - Compressor failed a blocking pager_get
VM - Compressor failed a blocking pager_get

Thread 0 Crashed:: bambustudio_main Dispatch queue: com.apple.main-thread
0   BambuStudio                   	       0x10177cbf3 append_menu_item(wxMenu*, int, wxString const&, wxString const&, std::__1::function<void (wxCommandEvent&)>, wxBitmap const&, wxEvtHandler*, std::__1::function<bool ()>, wxWindow*, int) + 115
1   BambuStudio                   	       0x10163df03 Slic3r::GUI::MenuFactory::append_menu_item_change_filament(wxMenu*) + 1283
2   BambuStudio                   	       0x10163d9e6 Slic3r::GUI::MenuFactory::object_menu() + 22
3   BambuStudio                   	       0x1015a0bea Slic3r::GUI::Plater::priv::on_right_click(Slic3r::GUI::Event<std::__1::pair<Eigen::Matrix<double, 2, 1, 2, 2, 1>, bool> >&) + 378
4   BambuStudio                   	       0x103542a1e wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 222
5   BambuStudio                   	       0x10354278b wxEvtHandler::ProcessEventLocally(wxEvent&) + 59
6   BambuStudio                   	       0x1035426a0 wxEvtHandler::ProcessEvent(wxEvent&) + 96
7   BambuStudio                   	       0x103542511 wxEvtHandler::ProcessPendingEvents() + 225
8   BambuStudio                   	       0x10349a2a7 wxAppConsoleBase::ProcessPendingEvents() + 119
9   BambuStudio                   	       0x10354c800 wxCFEventLoop::CommonModeObserverCallBack(__CFRunLoopObserver*, int) + 48
10  CoreFoundation                	    0x7ff800693550 __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 23
11  CoreFoundation                	    0x7ff8006933e2 __CFRunLoopDoObservers + 543
12  CoreFoundation                	    0x7ff800692874 __CFRunLoopRun + 840
13  CoreFoundation                	    0x7ff800691e6c CFRunLoopRunSpecific + 562
14  HIToolbox                     	    0x7ff8093405e6 RunCurrentEventLoopInMode + 292
15  HIToolbox                     	    0x7ff809340213 ReceiveNextEventCommon + 283
16  HIToolbox                     	    0x7ff8093400e5 _BlockUntilNextEventMatchingListInModeWithFilter + 70
17  AppKit                        	    0x7ff8030cbfad _DPSNextEvent + 927
18  AppKit                        	    0x7ff8030ca66a -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1394
19  AppKit                        	    0x7ff8030bcd19 -[NSApplication run] + 586
[... 548 lines cut ...]
Memory Module: BANK 0/ChannelA-DIMM0, 16 GB, DDR4, 2667 MHz, Micron, 16ATF2G64HZ-2G6E3
Memory Module: BANK 2/ChannelB-DIMM0, 16 GB, DDR4, 2667 MHz, Micron, 16ATF2G64HZ-2G6E3
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x7BF), wl0: Jul 12 2021 19:26:30 version 9.30.464.0.32.5.76 FWID 01-45ccefcd
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Ethernet, Ethernet, en0
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: APPLE HDD ST3000DM001, 3 TB
USB Device: G-DRIVE USB-C
USB Device: Mobile Drive
USB Device: USB31Bus
USB Device: USB3.1 Hub
USB Device: USB3.1 Hub
USB Device: USB3.0 Hub
USB Device: FaceTime HD Camera (Built-in)
USB Device: USB2.1 Hub
USB Device: SteelSeries Apex Pro
USB Device: USB2.1 Hub
USB Device: General Purpose USB Hub
USB Device: Arctis Pro Wireless
USB Device: Arctis Pro Wireless
USB Device: USB2.0 Hub
USB Device: USB Storage
USB Device: SteelSeries QCK Prism Cloth
USB Device: SteelSeries Rival 600
Thunderbolt Bus: iMac, Apple Inc., 47.5

bambu123, 2022-09-06:
@mapaolini 
Is it reproducible?

mapaolini, 2022-09-06:
It has only happened the one time so far.

parksddd, 2022-09-21:
@bambu123 Bambu studio crashes for me every time I save a project on mac. The file saves correctly, but crashes every time.

cmguo, 2022-09-22:
Please send us a new System Report. It can be fetched from Mac Console.

bambu123, 2022-10-09:
> @bambu123 Bambu studio crashes for me every time I save a project on mac. The file saves correctly, but crashes every time.

Same As #391

YifanWuBambu, 2022-10-14:
@mapaolini @parksddd 
Do you have this issue on v1.2.0.9?

YifanWuBambu, 2022-12-16:
This is fixed in v1.4.

closed 2022-12-16 by YifanWuBambu